prusa3d/PrusaSlicer #10312: sequential printing is not ok.

891paul, 2023-04-10:
### Description of the bug

When you start sequential printing two identically objects in the same coordinates on the y-axes. you can place them in one line with the distance of the radius. That's ok, but the height is it not. For example the height to the x-axes is 30mm the object height is 50mm and the two objects are - as said - in one line on y-coordinates, the x-axes will crash the finished object. That's a bug. 

It should also be possible, to definate the clear room as a rectangle ground size and not a circle around the nozzle, so that so nozzle isn't in the middle of the rectangle or circle. You spare more useable place then.

best regards

### Project file & How to reproduce

[prusa.zip](https://github.com/prusa3d/PrusaSlicer/files/11190442/prusa.zip)
pictures are out of alpha5, but alpha 6 is the same problem

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha6+win64-202303311519

### Operating system

win 10

### Printer model

BLV MGN CUBE

kubispe1, 2023-04-14:
Hi, thanks @lukasmatena  for fixing the issue. Complete individual objects and their height control will work similarly to 2.5.0. 
@freakydude there is already an issue, where to conversate arranging for Complete of individual objects #1052

closed 2023-04-14 by kubispe1

891paul, 2023-04-14:
sorry, but 2.5.0 is also not working. 

There you can't slice something in sequential mode. 
[image]
[image]

Height of the object is 53,09mm and clearence height setting is 30mm. Slicing not available. 
The idea of @freakydude is good, but height matters. 
You should place from the lowest to the tallest. Out of X/Y (0/0) to X max / Y0 and next starting X0 / Y+ (Nozzle to extruder left in front) + little security space. 
But this is only functionally, as long as the height of the printed object is lower than the clearance height. If it is taller, x-axis is crashing. So you have to start a new Y row as descripted further up. 

If the printed objects are heigher than the clearance height, you can't print them on the same Y line, because of the height you will crash. 
Sequential printing, as it is until now, is not useable.

891paul, 2023-05-17:
look at #10552