prusa3d/PrusaSlicer #10313: Bridge doesn't touch internal perimeter when extrusion width increased.

vovodroid, 2023-04-10:
### Description of the bug

When printing with such extrusion set combination:
[image]

bridge is printed in the air:
[image]

### Project file & How to reproduce

[Rail.zip](https://github.com/prusa3d/PrusaSlicer/files/11191245/Rail.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha6

### Operating system

Win10x64

### Printer model

Biqu B1 SE

Jan-Soustruznik, 2023-04-11:
Closed as duplicate issue - #10231

FidelCapo, 2023-04-19:
This is not duplicate of https://github.com/prusa3d/PrusaSlicer/issues/10231

vovodroid, 2023-05-14:
It looks better in beta-1, though I'm still not sure whether it's really correct now:

[image]

vovodroid, 2023-06-03:
Still bad in beta 4 with standard settings:

[image]
[Rail.zip](https://github.com/prusa3d/PrusaSlicer/files/11642021/Rail.zip)

vovodroid, 2024-03-21:
Looks good in 2.7.3-a1:

[image]

The same in 2.7.2:

[image]

Closing bug.

closed 2024-03-21 by vovodroid