prusa3d/PrusaSlicer #10314: [2.6.0alpha6] Retraction performs differently on layer change

Ohmarinus, 2023-04-10:
### Description of the bug

The problem is that stringing occurs *only* after a layer change move.

Stringing only occurs on the circled tower:
[image]

This is what it looks like in the print, notice how there's only stringing between the last tower of the previous layer and first tower of the next layer:
[image]

Here's the last segment highlighted before retraction and move to next pillar (next pillar is bottom one in screenshot):
[image]

The only other issue I can think of is that the travel direction has influence on how the 'thread' is broken between each pillar, but looking closely to the printer it doesn't look like that should matter. These are the directions highlighted (would be nice if the travel lines got an arrow of the vector to indicate the direction):
[image]

It also sounds like the machine is making a different sound on the layer change however, **the layer change should not really influence this as I'm using a Delta** which has the same feedrates set for XYZ. So there is no slowdown on a Z-raise.

I first asked to any ideas in the Prusa discord, we only found that `retract on layer change` was turned off, or set to `false` by default, which contradicts the info in the Prusa Knowledge base:
[image]

The description claims this value is set to false by default, but the article in the knowledge base claims this value should be left on:
[image]

So I'm assuming this should be set to `true` by default. However, turning on this value didn't make any impact to my print because on the layer height change, it also did a move and it would retract anyway.

Are there any other things I can change or try out or is this indeed an issue and unintended behavior of the slicer?

### Project file & How to reproduce

Here's the project file 3MF:
[retraction_on_layer_change_issue.zip](https://github.com/prusa3d/PrusaSlicer/files/11193839/retraction_on_layer_change_issue.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha6+win64

### Operating system

Win10

### Printer model

DIY Delta

Ohmarinus, 2023-04-11:
> You say the direction has no bearing but it should, that last part is the only one where the nozzle moves directly away from the seam into 'free air'. All the others the nozzle passes over the previous part.

Yes, normally I would agree, but I am utilizing Z-lift (0.2mm) so I would assume this Z-lift would prevent such an unwanted wipe.

neophyl, 2023-04-11:
Don't make any assumptions :)  
One thing I notice about that last travel move is that its higher than the ones between the other parts.  I think its the new layer height plus your z hop.  
As an experiment change your Lift Z to be 0.5 (which should match the one between start/end parts that is stringing).  See if it causes the strings between all the parts when set like that.  If it does then it would indicate that the lower 0.2 z hop is still wiping filament into the part as it does its travel which a higher z hop doesnt.

closed 2026-09-29 by prusaslicer-bot[bot]