prusa3d/PrusaSlicer #10315: Problem with comming out of sleep mode windows 11 laptop when busy with a slicing project.

Lonewolf2nd, 2023-04-11:
### Description of the bug

When I open a 3mf file and adjust certain things. I start printing. Normally I leave it open while printing small calibration tests.
But during the print my laptop goes into sleepmode when I wake up the laptop and want to resume and try to safe the 3mf project I get an internal error, which forces prusaslicer to close.
After this happens it wil happen everytime while saving a 3mf file, untill I reinstall prusa slicer.

https://forum.prusa3d.com/forum/prusaslicer/internal-error-after-coming-out-of-sleep-mode-windows-11/

### Project file & How to reproduce

Open 3mf file
Adjust something
Slice it
Let windows 11 go to sleep mode
Wake up laptop
Try to save project.

Screenshots of error in prusa forum post

https://forum.prusa3d.com/forum/prusaslicer/internal-error-after-coming-out-of-sleep-mode-windows-11/

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.1 and 2.5.2

### Operating system

Windows 11

### Printer model

Mk3s+

bubmann, 2023-04-11:
yeah my 2.6 alpha is most of the time just gone after coming out of sleep. and worst thing is, saving profiles does nothing, the new settings are still gone ... but that is probably a different issue

Lonewolf2nd, 2023-04-15:
Didn't try to safe profiles yet after bug. Going to let it crash again and try and see if that is an issue aswell. My 3mf file is saved when I select save prohect, even if it crashes.

closed 2026-09-29 by prusaslicer-bot[bot]