prusa3d/PrusaSlicer #10319: MacOS 2.6 Alpha6 Crash on project load

pwnpanda, 2023-04-11:
### Description of the bug

When loading projects, the application crashes fatally and reliably. See a few crash reports and 3mf files below. (Crash 2 is in comment below)

It seems to be related to loading projects with some kind of problematic setting enabled?
May also be related to #9871 .

Please let me know if there is anything else I can do to assist!

<details>
<summary>Case 1</summary>

File 1: 
[LED_Stand_v1.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11200879/LED_Stand_v1.3mf.zip)

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [14991]
Path:                  /private/var/folders/*/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.6.0-alpha6+MacOS-x64-202303311529 (PrusaSlicer-2.6.0-alpha6+MacOS-x64-202303311529)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-04-11 15:17:00.8004 +0200
OS Version:            macOS 13.3.1 (22E261)
Report Version:        12
Bridge OS Version:     7.4 (20P4252)
Anonymous UUID:        DCE9662F-A7A4-1E28-98EA-B95C1906DF15

Time Awake Since Boot: 770 seconds

System Integrity Protection: disabled

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00000000000000e8
Exception Codes:       0x0000000000000001, 0x00000000000000e8

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [14991]

VM Region Info: 0xe8 is not in any region.  Bytes before following region: 140737487945496
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      shared memory            7ffffff9c000-7ffffff9d000 [    4K] r-x/r-x SM=SHM  

Thread 0 Crashed:: slic3r_main Dispatch queue: com.apple.main-thread
0   PrusaSlicer                   	       0x10c87d5c2 Slic3r::_3MF_Importer::_load_model_from_file(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, Slic3r::Model&, Slic3r::DynamicPrintConfig&, Slic3r::ConfigSubstitutionContext&) + 5858
1   PrusaSlicer                   	       0x10c89c36c Slic3r::load_3mf(char const*, Slic3r::DynamicPrintConfig&, Slic3r::ConfigSubstitutionContext&, Slic3r::Model*, bool) + 940
2   PrusaSlicer                   	       0x10ca06d78 Slic3r::Model::read_from_archive(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, Slic3r::DynamicPrintConfig*, Slic3r::ConfigSubstitutionContext*, Slic3r::enum_bitmask<Slic3r::Model::LoadAttribute, void>) + 376
3   PrusaSlicer                   	       0x10d3a1b26 Slic3r::GUI::Plater::priv::load_files(std::__1::vector<boost::filesystem::path, std::__1::allocator<boost::filesystem::path>> const&, bool, bool, bool) + 3990
4   PrusaSlicer                   	       0x10d3b8b49 Slic3r::GUI::Plater::load_project(wxString const&) + 537
5   PrusaSlicer                   	       0x10d397a4e Slic3r::GUI::Plater::load_files(wxArrayString const&, bool) + 2814
6   PrusaSlicer                   	       0x10d396ef8 Slic3r::GUI::PlaterDropTarget::OnDropFiles(int, int, wxArrayString const&) + 120
7   PrusaSlicer                   	       0x10f11115f wxFileDropTarget::OnData(int, int, wxDragResult) + 63
8   PrusaSlicer                   	       0x10f2b7683 wxWidgetCocoaImpl::performDragOperation(void*, NSView*, void*) + 259
9   PrusaSlicer                   	       0x10f2b6c93 wxOSX_performDragOperation(objc_object*, objc_selector*, id<NSDraggingInfo>) + 51
10  AppKit                        	    0x7ff80a8ee334 NSCoreDragReceiveMessageProc + 672
11  HIServices                    	    0x7ff80ccf2cfb CallReceiveMessageCollectionWithMessage + 103
12  HIServices                    	    0x7ff80cced700 DoMultipartDropMessage + 105
13  HIServices                    	    0x7ff80cced4f6 DoDropMessage + 42
14  HIServices                    	    0x7ff80ccf0b58 CoreDragMessageHandler + 871
15  CoreFoundation                	    0x7ff8075f83ed __CFMessagePortPerform + 1107
[... 408 lines cut ...]

[image]

Test file 1:
[LED_Stand_v1.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11201085/LED_Stand_v1.3mf.zip)

Test file 2:
[LED_Stand_v1_low_support_detailed.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11201086/LED_Stand_v1_low_support_detailed.3mf.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha6+MacOS-x64

### Operating system

MacOs Ventura 13.3.1

### Printer model

Prusa Mini+

pwnpanda, 2023-04-11:
<details>
<summary>Case 2</summary>
[LED_Stand_v1_low_support_detailed.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11200924/LED_Stand_v1_low_support_detailed.3mf.zip)

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               PrusaSlicer [17234]
Path:                  /private/var/folders/*/PrusaSlicer.app/Contents/MacOS/PrusaSlicer
Identifier:            com.prusa3d.slic3r/
Version:               PrusaSlicer PrusaSlicer-2.6.0-alpha6+MacOS-x64-202303311529 (PrusaSlicer-2.6.0-alpha6+MacOS-x64-202303311529)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2023-04-11 15:32:18.4293 +0200
OS Version:            macOS 13.3.1 (22E261)
Report Version:        12
Bridge OS Version:     7.4 (20P4252)
Anonymous UUID:        DCE9662F-A7A4-1E28-98EA-B95C1906DF15

Time Awake Since Boot: 1600 seconds

System Integrity Protection: disabled

Crashed Thread:        0  slic3r_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x00000000000000e8
Exception Codes:       0x0000000000000001, 0x00000000000000e8

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [17234]

VM Region Info: 0xe8 is not in any region.  Bytes before following region: 140737487945496
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      shared memory            7ffffff9c000-7ffffff9d000 [    4K] r-x/r-x SM=SHM  

Thread 0 Crashed:: slic3r_main Dispatch queue: com.apple.main-thread
0   PrusaSlicer                   	       0x1098145c2 Slic3r::_3MF_Importer::_load_model_from_file(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, Slic3r::Model&, Slic3r::DynamicPrintConfig&, Slic3r::ConfigSubstitutionContext&) + 5858
1   PrusaSlicer                   	       0x10983336c Slic3r::load_3mf(char const*, Slic3r::DynamicPrintConfig&, Slic3r::ConfigSubstitutionContext&, Slic3r::Model*, bool) + 940
2   PrusaSlicer                   	       0x10999dd78 Slic3r::Model::read_from_archive(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>> const&, Slic3r::DynamicPrintConfig*, Slic3r::ConfigSubstitutionContext*, Slic3r::enum_bitmask<Slic3r::Model::LoadAttribute, void>) + 376
3   PrusaSlicer                   	       0x10a338b26 Slic3r::GUI::Plater::priv::load_files(std::__1::vector<boost::filesystem::path, std::__1::allocator<boost::filesystem::path>> const&, bool, bool, bool) + 3990
4   PrusaSlicer                   	       0x10a34fb49 Slic3r::GUI::Plater::load_project(wxString const&) + 537
5   PrusaSlicer                   	       0x10a32ea4e Slic3r::GUI::Plater::load_files(wxArrayString const&, bool) + 2814
6   PrusaSlicer                   	       0x10a32def8 Slic3r::GUI::PlaterDropTarget::OnDropFiles(int, int, wxArrayString const&) + 120
7   PrusaSlicer                   	       0x10c0a815f wxFileDropTarget::OnData(int, int, wxDragResult) + 63
8   PrusaSlicer                   	       0x10c24e683 wxWidgetCocoaImpl::performDragOperation(void*, NSView*, void*) + 259
9   PrusaSlicer                   	       0x10c24dc93 wxOSX_performDragOperation(objc_object*, objc_selector*, id<NSDraggingInfo>) + 51
10  AppKit                        	    0x7ff80a8ee334 NSCoreDragReceiveMessageProc + 672
11  HIServices                    	    0x7ff80ccf2cfb CallReceiveMessageCollectionWithMessage + 103
12  HIServices                    	    0x7ff80cced700 DoMultipartDropMessage + 105
13  HIServices                    	    0x7ff80cced4f6 DoDropMessage + 42
14  HIServices                    	    0x7ff80ccf0b58 CoreDragMessageHandler + 871
15  CoreFoundation                	    0x7ff8075f83ed __CFMessagePortPerform + 1107
16  CoreFoundation                	    0x7ff8075565a3 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE1_PERFORM_FUNCTION__ + 41
17  CoreFoundation                	    0x7ff8075564e3 __CFRunLoopDoSource1 + 540
18  CoreFoundation                	    0x7ff807555161 __CFRunLoopRun + 2694
19  CoreFoundation                	    0x7ff807554071 CFRunLoopRunSpecific + 560
20  HIToolbox                     	    0x7ff810fbcfcd RunCurrentEventLoopInMode + 292
21  HIToolbox                     	    0x7ff810fbcdde ReceiveNextEventCommon + 657
22  HIToolbox                     	    0x7ff810fbcb38 _BlockUntilNextEventMatchingListInModeWithFilter + 64
23  AppKit                        	    0x7ff80a5e67a0 _DPSNextEvent + 858
24  AppKit                        	    0x7ff80a5e564a -[NSApplication(NSEvent) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 1214
25  AppKit                        	    0x7ff80a5d7cb8 -[NSApplication run] + 586
26  PrusaSlicer                   	       0x10c2318ba wxGUIEventLoop::OSXDoRun() + 186
[... 333 lines cut ...]
  "trialInfo" : {
  "rollouts" : [
    {
      "rolloutId" : "5f72dc58705eff005a46b3a9",
      "factorPackIds" : {

      },
      "deploymentId" : 240000015
    },
    {
      "rolloutId" : "6112d17137f5d11121dcd4e2",
      "factorPackIds" : {

      },
      "deploymentId" : 240000250
    }
  ],
  "experiments" : [

  ]
}
}

```
</details>

kubispe1, 2023-04-17:
Thanks we will look into it. 
SPE-1662

closed 2023-04-24 by FidelCapo

pwnpanda, 2023-04-24:
@FidelCapo Thanks! 👍