prusa3d/PrusaSlicer #10321: Allow Printing ONLY Intersection of Modifier with Object (like COMPLEMENT of NEGATIVE MODIFIER)

puterboy, 2023-04-11:
Often when I am testing a new design, I want to print out only a small region of the final print.
Typically, this requires a sequence of horizontal & vertical cuts to remove all the areas I don't want to print.

However, it would be easier if I could add a modifier in the shape of say a box or cylinder, overlap the modifier with the volume I want to print and then have it ONLY print the overlap volume (i.e, cut away all areas NOT overlapped by the modifier)

This would be the Boolean equivalent of printing only the complement of the intersection of the modifier and the object.

This could presumably be added as an option to a negative modifier to print the complement of the volume spanned by the negative modifier.

roaldarbol, 2023-05-09:
This would be such a useful feature for quick, easy prototyping. Would love to see it implemented too!

darthgloda, 2023-06-06:
This would be really useful.

machinekoder, 2023-12-05:
Yes, that would be very useful. I started to use the "positive" and "negative" volumes a lot, and sometimes an "intersection" volume would be ideal.

Can maybe someone of the maintainers point me to where the negative/additive volume feature is implemented, so I can take a stab at it?

puterboy, 2023-12-11:
Any possibility of adding this?

To emphasize when I am test printing an object, I typically only want to print a portion of the total object to save both time and material.
Without an intersection modifier, I have to create a complex combination of "Negative Volumes" and/or "Cuts" in order to carve out the small section that I want to print.
It would be INFINITELY easier to just intersect the desired area with a box or cyclinder (or other default shape)...
This to me is MORE useful and important than a negative volume.

jmickelin, 2024-04-06:
One "workaround" is to create an impromptu "cutting tool" STL within Prusa Slicer by first creating a box that covers your entire model, then adding a negative volume to this box that covers just the part that you want to print. Then you can export this as an STL and then immediately `Load` it as a negative volume modifier for your actual model.

It still sucks, especially since the UI makes it impossible to line things up (even "invisible" objects are rendered opaque, only in gray – the most invisible color :smirk:). So effectively you have to to do a lot of trial-and-error with a blindfold on and checking if you hit the right area by clicking `Slice`. But it beats creating 6 cutting planes and having to individually select whether to keep part A or part B for each of them. It's also somewhat repeatable, and quick to apply again if you (say) reload your model from disk.

And most definitely beats importing the model in your CAD software, jumping through all the steps for converting it from mesh to a solid, just to intersect it and export it again. Something that convolutes a routine task that should wholly be the concern of the slicing-and-printing process with that of the modelling one.

puterboy, 2024-11-18:
Devs PLEASE add this feature - it is so simple yet so powerful.
Today I only wanted to print the center of a print. It would be trivial to do this by intersecting a cylinder with my print. There is no easy way to do this without such an intersection feature!!!

Laltani, 2025-05-21:
This feature would make my life so much easier, for the reasons mentioned by everyone else.

puterboy, 2025-05-21:
Why hasn't this totally obvious feature been added yet????

puterboy, 2025-05-27:
It has now been **more than 2 years!!!!** yet none of the devs have even responded to this feature request to let us know they are reading this let alone listening to all the positive feedback about this request

ihor-drachuk, 2025-06-05:
Still needed

puterboy, 2026-09-02:
Keep this alive until fixed!
This is a much desired feature and it has been IGNORED for years!