prusa3d/PrusaSlicer #10330: New Filament setting for "Wipe/Probe Temperature."

ScottWell1, 2023-04-11:
Add a new filament property for "Wipe/Probe Temperature".

The Mini, MK4 and XL models now perform a "wipe" of the nozzle and "probe" directly on the bed using load-cell tech. For that wipe and probe to succeed, the filament needs to be at a temperature where the filament is very soft, but not fully liquified to prevent ooze.  This temperature is currently not represented as a data element for each filament.

The workaround currently used is implemented in the Prusa-supplied Printer profiles as part of the Printer Start Gcode, using If-then-elsif-else logic such as this:
```
{if filament_type[initial_tool]=="PC" or filament_type[initial_tool]=="NYLON"}
M104 S{first_layer_temperature[initial_tool]-25} ; set extruder temp for bed leveling
M109 R{first_layer_temperature[initial_tool]-25} ; wait for temp
{elsif filament_type[initial_tool]=="FLEX"}
M104 S210 ; set extruder temp for bed leveling
M109 R210 ; wait for temp
{else}
M104 S170 ; set extruder temp for bed leveling
M109 R170 ; wait for temp
{endif}
```
Note that the above uses a default of 170, however there are cases (aside from Nylon and Flex) where more if-then-else logic will need to be added, further complicating the code.  And in any case, this ties a filament-related parameter to the Printer instead of the Filament.

Having the wipe/probe temperature as a data element tied to the filament could simplify the Start Gcode above to something like this:
```
M104 S{filament_wipe_temperature[initial_tool]}
M109 R{filament_wipe_temperature[initial_tool]}
```

zzjlamb, 2023-08-19:
I support this request. I am getting tiny dots of plastic deposited on the build plate during probing with some brands of PLA which have a lower melting point. For those filaments I would like to set a lower probing temperature. It makes sense for this to be a filament setting instead of custom gcode in the printer settings.

swannet, 2023-10-25:
I support this request too, for the same reason. I print a lot of Sunlu PLA Meta (at 185/180 printing temp) so 170 for nozzle cleaning is to high because it keeps oozing at 170 so Nozzle cleaning failed (only with this filament). So I also approve the suggestion to make it filament setting.

pacaj2am, 2024-01-03:
I had same issues with Filamentum Woden filaments as well LW PLAs, would be great to set that as param of filament

justin-robinson, 2024-02-25:
This would help with sheets melting. My PET sheet melts over 100C so it has craters melted into it where the nozzle probes.

VorpalBlade, 2024-05-20:
I'm having some Overture TPU that PrusaSlicer by default tries to do nozzle cleaning and probing with at 210 C. It oozes like crazy at that temperature, and the nozzle cleaning *always* fails. I have to lower the temp to 200 C for reliable operation. Yet looking at the start G-code that is handled on a per-machine basis, with an option for `first_layer_temperature[0] - 10`, However since the first layer temperature of this filament is 230 C that is completely useless.

VorpalBlade, 2026-09-18:
From what I can see, PrusaSlicer 3 doesn't have this.