prusa3d/PrusaSlicer #10331: Fan GCode wrong

TravisWilder, 2023-04-12:
### Description of the bug

Fan Speed shows in Slice View as 100% 
in Gcode there is a wrong M106 Command:

M106 S53.55

It seem the values are not converted to the 0-255 range
Fan Settings:
Allways - no auto - Min 20% => Works
Allways - no auto - Min 21% (and higher) => errors in gcode

### Project file & How to reproduce

[Archiv.zip](https://github.com/prusa3d/PrusaSlicer/files/11210325/Archiv.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6+1-lm-mailsail-fix-MacOS-arm64

### Operating system

Mac Ventura 13.3.1

### Printer model

Voron V0

TravisWilder, 2023-04-12:
similar as #9331

TravisWilder, 2023-05-17:
Hello
similar to #10542 and #10380 

also in Beta 2 the slicer generates GCODE like

M106 S158.1
which is shown as 100% by the the slicer in the GUI

TravisWilder, 2023-05-17:
WipeTower.cpp : 406 : m_gcode += "M106 S" + std::to_string(unsigned(255.0 * speed / 100.0)) + "\n";
GCodeWriter.cpp; 528 : code << "M106 S" << 255.0 * speed / 100.0; break;
This seem to result floats and not INTs

TravisWilder, 2023-05-30:
It is still the case for me in Beta 3
Fan Speed is shown 0 or 100 in preview
Gcode produced include ... M106 S45.9

lukasmatena, 2023-06-13:
@TravisWilder We were not able to reproduce your issue so far. But I noticed, that the people reporting similar issues (#10380, maybe #10543) are using some non-English translation (German, French, Russian). If the parsing of floating point values is wrong, it may point to some incorrectly set locales.

Could you please set PrusaSlicer to English and check whether you are still having the problem?

Thanks. In the meantime, I will prepare a testing build for you where the parsing of the number will be reworked.

TravisWilder, 2023-06-13:
> @TravisWilder We were not able to reproduce your issue so far. But I noticed, that the people reporting similar issues (#10380, maybe #10543) are using some non-English translation (German, French, [...]

Hello and Thanks @lukasmatena  to have a look:

you pointed in the right direction:

I tried PS2.6RC1 on a 
MacOS in English & PS in English> Worked
MacOS in German & PS in German > Fails
MacOS in German & PS in English > Works

So it seems a language issue in PS, where PS switches how float are used
In German the "." is "," for float. So 31.24 % is 31,24 % and the . might be used for 1.000 in German

lukasmatena, 2023-06-13:
Thanks. Could you please give it a try in the following build? It is 2.6.0-rc1 plus [this commit](https://github.com/prusa3d/PrusaSlicer/commit/d492fc0efbff2758680f6ad310eb8dfd7c20c54c).

[PrusaSlicer-2.6.0-rc1+1-lm-fan-speed-bug-investigation-linux-x64-GTK3-gd492fc0-202306130717.AppImage](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.6.0-rc1+1-lm-fan-speed-bug-investigation-linux-x64-GTK3-gd492fc0-202306130717.AppImage)
[PrusaSlicer-2.6.0-rc1+1-lm-fan-speed-bug-investigation-MacOS-universal-gd492fc0ef-202306130729.dmg](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.6.0-rc1+1-lm-fan-speed-bug-investigation-MacOS-universal-gd492fc0ef-202306130729.dmg)
[PrusaSlicer-2.6.0-rc1+1-lm-fan-speed-bug-investigation-win64-gd492fc0ef-202306130718.zip](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.6.0-rc1+1-lm-fan-speed-bug-investigation-win64-gd492fc0ef-202306130718.zip)

TravisWilder, 2023-06-13:
works for me now with the rc plus build

lukasmatena, 2023-06-14:
Should be fixed in [2.6.0-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-rc2). Closing.

@TravisWilder Thanks for bugging me enough to not let it fall through.

closed 2023-06-14 by lukasmatena