prusa3d/PrusaSlicer #10332: Slower auto arrange and slicing with 2.6 compared to 2.5

BMacours, 2023-04-12:
### Description of the bug

Tested with 64 small objects : Auto arrange and slicing take approx 2x times longer with Prusaslicer 2.6 than with 2.5

### Project file & How to reproduce

[Test-2.6-SlowerArrangeAndSlice.zip](https://github.com/prusa3d/PrusaSlicer/files/11210999/Test-2.6-SlowerArrangeAndSlice.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha6

### Operating system

win64

### Printer model

Mini

foreachthing, 2023-04-14:
That's a pure torture test! 😀 
Well, initially I had "enable rotation" enabled. To disable that option helped a lot! 
But, yes, it takes a bit longer: PS2.5 @ 100% -> PS2.6  @ 75%. Slicing is about the same as arrange where 2.5 takes the cake (2.6 @ 90%).
Also noticed that the progress bar doesn't update if PS is not the active window.

All on Windows 10 x64, affinity changed to give each instance 50% of the available CPUs.

BMacours, 2023-04-14:
Yes I like torturing things 😀 But it isn't an academical case : I'm used to have such projects for printing slot car parts.

closed 2026-09-29 by prusaslicer-bot[bot]