prusa3d/PrusaSlicer #10335: [PrusaSlicer 2.6.0 alpha6] new solid infill not optimal for small spaces and causing tiny wipe moves

cordilon, 2023-04-12:
### Description of the bug

After two failed prints of the friendly slug, I took a closer look at the sliced file and noticed that the new circular solid infill method might be the main problem:
In small and/or circular areas the last bit of the circular solid infill can be a very short extrusion move (barely larger than the nozzle diameter), which isn't really needed but might cause problems due to rapid retraction.
The following wipe is limited to that short extrusion and is therefore not only useless, but will then pull up the print while performing Z-hop, which is especially problematic on overhangs.

[image]

I'd prefer the old solid infill method here or more settings to tune the circular infill!

### Project file & How to reproduce

[SlugV1.1_scaled_[x2.25]_pt15_slic3r-project_PS-2.6.0a.zip](https://github.com/prusa3d/PrusaSlicer/files/11214188/SlugV1.1_scaled_.x2.25._pt15_slic3r-project_PS-2.6.0a.zip)

Warning: The printer is a modified Prusa i3 Mk3 with Bondtech Extruder and custom start and end gcode!

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 alpha6

### Operating system

Windows 10 64bit

### Printer model

Prusa i3 Mk3 (modified with Bondtech Extruder etc.)

kubispe1, 2023-04-13:
Thank you for your experience. 
Solid infill for smaller areas is implemented to Ensure vertical shell thickness. There is an issue already  #10202. Please conversate there. I am closing this issue, Thanks

cordilon, 2023-04-13:
@kubispe1 thanks, but issue #10202 is not really about the same problem. What concerns me the most is the short extrusion and the behavior of the wipe move.

kubispe1, 2023-04-17:
OK

closed 2026-09-29 by prusaslicer-bot[bot]