prusa3d/PrusaSlicer #10337: Problem opening stl from dolphin

DylanMS, 2023-04-12:
### Description of the bug

When I try to open STL file from dolphin file manager my desktop gose black KDE shell die only restart help.
I had 2.6.1 version of appimage and openit without problem when replace appimage with newer appimage it stop working old also. 

PS- I named new application PrusaSlicer.AppImage old app named PrusaSlicer-old.AppImage and also put in same location like old app. Shortcut from desktop work. 

### Project file & How to reproduce

All STL files is problem opening from desktop file manager dolphin. I can open from application stl file import or 

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.1

### Operating system

openSuSe 15.4

### Printer model

00

Jan-Soustruznik, 2023-04-13:
Hello, @DylanMS , 

could you re-check the version number of slicer, because 2.6.1 isn't released yet. Last released final version is 2.5.2 or Alpha 2.6.0-Alpha6 ... 

Anyway, problem is .. if you try to open the stl file from the manager by double-click and select the app, then manager window goes black, correct?

Thank you, have a nice day

DylanMS, 2023-04-13:
Hello sorry I mean 2.5.1 and newer 2.5.2. All desktop kde are black not responsive. I can access console with ctrl+alt+f4. Thank you

Jan-Soustruznik, 2023-04-13:
@DylanMS, 

Do you have same problem with 2.5.0? Or can you try the latest alpha6? https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6

DylanMS, 2023-04-14:
Hello i now tested 2.5.0 and it working. So again switch to newer app 2.5.2 and not working. So i was delete Association file STL  in KDE Settings/Application/ search for .stl association remove it and save and add new association of file again save. Now working. I did not try 2.6.0.

BUT... when i try to do association from PrusaSlicer menu in application Konfigurace/Integrace do systému (using CZE language I dont know English name for this option...) it again problem with not responding black desktop. I hope you understand.

closed 2026-09-29 by prusaslicer-bot[bot]