prusa3d/PrusaSlicer #10339: Bug in latest Firmware MK3S+

MZimbar, 2023-04-12:
### Description of the bug

When I send print jobs to my Prusa MK3S+, I always get a message that says:

G-code sliced for a different printer type.  Continue?

### Project file & How to reproduce

I downloaded the firmware a second time just to make sure I didn't get the wrong firmware but i definitely had the right one.

This is the link i used for my MK3S+
https://cdn.prusa3d.com/downloads/firmware/prusa3d_fw_3_12_2_MK3S.zip#_ga=2.143661618.645058281.1681262595-1143676791.1672873255

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.2

### Operating system

Windows 11

### Printer model

Prusa MK3S+

sarusani, 2023-04-13:
Try:
https://github.com/prusa3d/Prusa-Firmware/releases/download/v3.12.2/prusa3d_fw_MK3S_MK3S+_3_12_2_5713.hex

Works fine for me.

rtyr, 2023-04-13:
You are likely using wrong printer profile for slicing (it can also happen if you open 3MF created by someone else). If the problem persists, please contact our tech-support.

closed 2023-04-13 by rtyr

MZimbar, 2023-04-14:
Profile in Prusa Slicer is set to Original Prusa i3 MK3S & MK3S+.  This is getting really annoying.  I re-applied the latest firmware and the first job went though without the message.  Sent the same job again and the message is back.

MZimbar, 2023-04-15:
> Try: https://github.com/prusa3d/Prusa-Firmware/releases/download/v3.12.2/prusa3d_fw_MK3S_MK3S+_3_12_2_5713.hex
> 
> Works fine for me.
> 
> If you're still getting the error, make sure you have the correct printer set in PrusaSlicer.

This firmware did the trick.  Must be something wrong with the one included in the following zip file:  https://cdn.prusa3d.com/downloads/firmware/prusa3d_fw_3_12_2_MK3S.zip#_ga=2.241095811.1232184667.1681515780-1143676791.1672873255

rtyr, 2023-04-15:
> This firmware did the trick. Must be something wrong with the one included in the following zip file: https://cdn.prusa3d.com/downloads/firmware/prusa3d_fw_3_12_2_MK3S.zip
> 
> https://github.com/prusa3d/Prusa-Firmware/releases/download/v3.12.2/prusa3d_fw_MK3S_MK3S+_3_12_2_5713.hex

Both of these files seems to be identical.

MZimbar, 2023-04-15:
> Both of these files seems to be identical.

They appear identical but I downloaded it twice and flashed it twice.  Another thing I noticed with the zip file version was that Autoload filament was disabled.  Every time i unloaded filament I had to go into the menu and select load filament.  That was new behaviour.

Maybe somebody updated the zip file.  Glad it is all working as my Revo 6 just arrived and want to install that soon.