prusa3d/PrusaSlicer #10344: Remember 'display seam' setting

Renohedron, 2023-04-13:
Could we please have 'display seam' setting being remembered, ideally forever or by default? Pretty pleeease?
Seriously, seam is THE MOST result affecting feature with the modern precise printers, we need to be always aware of it's location. And it's very annoying to always enable it. You probably won't want being remembered only this specific sliced view setting but it's the most important of those.

[image]

foreachthing, 2023-04-14:
Dup: https://github.com/prusa3d/PrusaSlicer/issues/9468

Jan-Soustruznik, 2023-04-14:
Issue closed as duplicate https://github.com/prusa3d/PrusaSlicer/issues/9468

closed 2023-04-14 by Jan-Soustruznik