prusa3d/PrusaSlicer #10346: Single Top/ Bottom Perimeter

TjardF, 2023-04-14:
**Is your feature request related to a problem? Please describe.**
On surfaces with many features on tight space you get problems with too many Perimeters.
They create ugly gaps between the individual perimeters.
Printing just one Perimeter on Top/ Bottom (selectable maybe with changeable width) would gain lots aesthetic without getting any weaknesses.

**Describe the solution you'd like**
Maybe like seen on SuperSlicer:
Checkbox:
Top Perimeter
Bottom Perimeter
Numeric Input:
Line width

These create the wanted effect without other changes in the slicing algorithm.

**Describe how it would work**
the slicing options need to be changed for these layers, shouldn't be that hard in my mind, but didn't looked into PrSl code....

**Describe alternatives you've considered**
Just a On/Off thing without these additional parameters maybe?

**Additional context**
Just take a look at these Examples.
I used an old Version of SuperSlicer to show the function of the feature.
Hope its understandable....

[image]
[image]
[image]

Due to PA and plastic flow in reality these gaps tend to be a bit annoying for me, maybe I'm not alone

sarusani, 2023-04-14:
You can add an height modifier (or two for this particular case) and change the amount of parameters for the bottom and top layers that way.

rtyr, 2023-04-14:
duplicate of https://github.com/prusa3d/PrusaSlicer/issues/7986.

closed 2023-04-14 by rtyr

dvomang, 2023-04-22:
Can we reopen this? It seems like something so simple to add that would make a HUGE difference. Looks like the other thread #7986 was also closed.

rtyr, 2023-04-22:
7986 is open