prusa3d/PrusaSlicer #10348: no retract after overhang print

SphaeroX, 2023-04-14:
### Description of the bug

After the overhangs are printed, no retract takes place, see the video here: https://youtu.be/ZjUfUGnsxck

### Project file & How to reproduce

[retract.zip](https://github.com/prusa3d/PrusaSlicer/files/11230540/retract.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha6+win64-202303311519

### Operating system

Win1

### Printer model

custom

lukasmatena, 2023-04-14:
You have `Print Settings->Infill->Advanced->Only retract when crossing perimeters` active. That is quite likely the culprit. Could you please check with that option disabled? Thanks.

SphaeroX, 2023-04-14:
Oh man sorry it was really activated after I deactivated it was good again, thanks!

closed 2023-04-14 by SphaeroX