prusa3d/PrusaSlicer #10349: Anomalies occur in rafts when Snug is used for support.

viper0078, 2023-04-14:
### Description of the bug

If Snug is used for support, the top surface of the raft cannot be formed because the printing angle of the top surface is the same as the first layer.

[image]

### Project file & How to reproduce

[box 10x3.zip](https://github.com/prusa3d/PrusaSlicer/files/11231604/box.10x3.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5

### Operating system

win11

### Printer model

snapmaker2.0

lukasmatena, 2023-04-14:
@viper0078 Can you please try it in [2.6.0-alpha6](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha6)? Thanks.

viper0078, 2023-04-14:
Thank you. I hope to have an official release soon.

closed 2023-04-14 by viper0078