prusa3d/PrusaSlicer #10353: Ironing doesn't care about ironed surface, dragging the nozzle on it, ruining the whole thing

Goury, 2023-04-14:
### Description of the bug

Ironing's supposed to produce smooth surfaces.
Moving over already ironed surface must not be allowed.
Or, at the very least (if this is too hard for you to fix), allow turning nozzle lifting for ironable surfaces.

### Project file & How to reproduce

1. Turn ironing on
2. Travel moves are generated where ironing took place

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5, 2.6.0-alpha6

### Operating system

Ubuntu 20.04

### Printer model

Irrelevant

rtyr, 2023-04-14:
Enable z-hop

bubnikv, 2023-04-14:
solved, closing

closed 2023-04-14 by bubnikv

Goury, 2023-04-14:
What the actually fuck?  
Now you aren't even trying to pretend you're not a lazy asshole?

Z hop is a sure way to ruin a print by stringing and many other issues coming along.  
And there is no way to turn it on just for ironing, it's either fuck you in general or fuck your horizontal surfaces in particular.

You didn't solve anything, you've just closed an issue because you don't want to do your job.

foreachthing, 2023-04-14:
@Goury, if you're here to post an issue and/or need help, I'd suggest to refrain from profanities. Prusa is not required to do anything (for free, anyways). Don't like it, don't use it.
I wouldn't be happy with this answer either but you can still take matters in your own hands and PR _your_ solution to the issue.

ciriousjoker, 2024-07-01:
Imo this should be reopened. I don't agree with Goury's language, but the issue isn't really solved either way.

- I guess you could argue that this is a duplicate of #5585, at least it would be solved if this was implemented