prusa3d/PrusaSlicer #10357: Recursive Gap Fill

CCS86, 2023-04-14:
I know that many people think that Arachne is "the way forward" but I am not so convinced. I find many examples where it does not work optimally. I think that refining the classic perimeter generator can provide better overall results.

The biggest room for improvement in my eyes, would be to add "recursive gap fill" and disallow "perimeter overlap" which is currently  allowed (correct me if I am wrong) 80% overlap in external perimeters and 20% in other perimeters. These overlap amounts are not flow corrected. So, in areas where there are long runs of parallel perimeters, can cause massive over-extrusion problems.

The classic perimeter generator must work in complete loops (or pairs of extrusions). So, if we disallow un-compensated overlap, we inevitably find ourselves with a gap that is just shy of double the defined extrusion width:

[image]

Then, the current gap fill behavior is to fill that with a single extrusion:

[image]

I think it is relatively obvious that this will not be a successful extrusion. Even if capped by maximum volumetric rate, the kinematics of trying to flow plastic rapidly into that form factor just won't work. You will most likely under-extrude that bead, failing to bond on either side, cause excessive nozzle pressure, etc.

Instead, if we define a maximum gap fill extrusion width, and ensure that the gapfill is appropriately adjacent to the last perimeter line, we can fill every gap at, or below, this maximum gap fill width:

[image]

Then, if we run the gap filling algorithm one more time, we will get all remaining gaps:

[image]

This seems like the easiest path towards full control of extrusion width, while filling all gaps, and with no defined over-extrusion to "cheat" our way there.

CCS86, 2023-06-08:
Boy, I was at least expecting some discussion on this request. :/

CCS86, 2023-06-24:
Maybe now that 2.6.0 is stable, this feature can get looked at.

CCS86, 2023-07-09:
Bueller?

CCS86, 2023-08-23:
4 months and no reply. 👎

CCS86, 2023-10-03:
?

CCS86, 2023-11-10:
?

jueank, 2023-11-21:
I'd vote for some of the devs doing a prototypical implementation. :) ;)

CCS86, 2023-12-01:
Crickets

CCS86, 2024-03-14:
Man, the PS devs are unresponsive to so many issues / requests. 

Even just dropping a comment to acknowledge reading the issue would be nice.

closed 2026-09-29 by prusaslicer-bot[bot]