prusa3d/PrusaSlicer #10358: Pressure Equalizer Causing Layer Misalignment

dmbaio, 2023-04-15:
### Description of the bug

When using the volumetric slope settings for the pressure equalizer, combined with the color change g-code below (to raise Z before and lower Z after color change because the AnkerMake M5 does not support M600 parameters yet), the Z values for the rest of the layer that the color change occurs on are corrupted. Setting the pressure equalizer settings to 0 eliminates the issue.

[image]

```
;COLOR CHANGE
G91
G1 Z30 F600
G90
M600
G91
G1 E-3 F1200
G1 Z-30 F600
G90
```

[image]

### Project file & How to reproduce

[Color Change Test.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11238156/Color.Change.Test.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha6+MacOS-arm64

### Operating system

macOS Ventura 13.4 Beta (22F5037d)

### Printer model

AnkerMake M5

kubispe1, 2023-04-20:
You are right. This is probably an issue of the Preasure equalizer. Switching from absolute to relative coordinates is not supported by the Preasure equalizer.

dmbaio, 2023-04-23:
@kubispe1, I was playing around with alternative color-change g-codes to work around the issue of having to disable pressure equalizer, and I tried this (just as a test, so I wasn't including things like retractions, etc.):

```
;COLOR CHANGE
G0 X0 Y0 F1800
M400
G0 Z100 F900
M600
```

This avoids any absolute/relative positioning mode changes and doesn't use any relative coordinates at all, but the issue still occurs. What is even more interesting is that with this color_change_gcode block, **_disabling pressure equalizer does not fix the issue._** With this code and pressure equalizer disabled, the issue is still there. This would lead me to think that the issue must not be directly caused by pressure equalizer after all.

closed 2026-09-29 by prusaslicer-bot[bot]