prusa3d/PrusaSlicer #10364: Allow custom variable in Filament Settings to be accessed in Printer Settings

CodeMonkeyX, 2023-04-15:
I would like to be able to access global variables defined in "Filament Settings -> Custom G-Code" in "Printer Settings -> Custom G-Code"

I am I trying to define a Chamber Temperature that is defined in the Filament Settings. What I did was add a global variable in the Filament Settings Start G-Code. But it looks like that gets parsed after the Printer Settings Start G-Code so the Printer Settings does not see it for the Print_Start macros.

Maybe a new text box in the "Filament Settings -> Custom G-Code" page that just says "Global Config Variables" then we can add any custom G-Code variables we might need for custom macros, that might not be available in the default user interface. 

Then we could manually define anything related to the filament, like: 
Chamber Temp
Filament Color

Actually I was just looking at Super Slicer and they have this in the "Notes" page of the Filament Settings under "Custom variables."

bubnikv, 2023-04-17:
> I'm interested in the same feature. Want to set a pressure advance in my
> filament settings an pass them to a klipper START_PRINT gcode macro from
> my printers start gcode.

In PrusaSlicer 2.6.0-alpha6 you may declare a global variable in filament
start G-code and then access it by the printer settings custom G-code
blocks, not exactly by the start G-code though.

CodeMonkeyX, 2023-04-17:
@bubnikv Thank you for the reply. Maybe it's a bug in alpha6 then? Because that is the reason I was trying this, I got excited when variables were added to alpha6 and this was the first thing I tried.

Maybe I am doing it incorrectly, because I am not familiar with the scripting language used. Here is what I tried to do.

_Filament Setting -> Start G-Code_
```
{
global chamber_temp = 60;
}
```

_Printer Setting -> Start G-Code_
```
print_start EXTRUDER=[first_layer_temperature[initial_tool]] BED=[first_layer_bed_temperature] CHAMBER={chamber_temp}
```
Note: I have tried both curly and square brackets.

When processing a G-Code file I get the error `invalid custom code`
```
start_gcode
Parsing error at line 2: Variable does not exist
print_start EXTRUDER=[first_layer_temperature[initial_tool]] BED=[first_layer_bed_temperature] FILAMENT_TYPE=[filament_type] CHAMBER={chamber_temp}
```

I did a test and just put some output in the Filament G-Code  section and it seems to get inserted after the Printer G-Code section. So I am wondering if the variable is just not created yet when the Printer G-Code section is processed?

HitLuca, 2023-10-05:
This is one of the few things that keeps me back from switching to prusaslicer. Having filament-specific variable helps with setting chamber temps (which isn't available in PS but is an actual setting in SS), and most importantly pressure advance values by filament brand

EverlastEngineering, 2024-01-21:
Here's how I used custom variables for configurable extrusion flow per layer. It might be of use for some here. This example sets the extrusion flow to 105% by using the `M221` command.

In the `Filament Settings -> Custom G-code -> Start G-code`, add a global variable:

```
{global first_layer_filament_extrusion_factor = 1.05;}
```

Then in `Printer Settings -> Custom G-code -> Before layer change G-code`, add:

```
M221 S{if layer_num==0}{first_layer_filament_extrusion_factor * 100}
{else}100
{endif} 
```
I dislike the `M221` command and others like it, because once upon a time I worked for hours setting up a very delicate print only to find I'd stopped a print earlier with the `M221` set to something other than 100, and my older startup code didn't have that as a parameter to restore to 100, so I printed with the wrong extrusion rate for the whole (ruined) print. 

So, consider adding `M221 S100` to your Startup G-code to be safe.

DasLukas, 2024-01-21:
In the meantime, I was no longer reliant on the global variable, but thank you nonetheless for the detailed explanation.

CodeMonkeyX, 2024-01-21:
Not really a solution but I started using Orca recently. It has chamber temps, pressure advance settings and other features useful to non Prusa machines built in.

Sent from [Proton Mail](https://proton.me/mail/home) for iOS

Jan-Soustruznik, 2024-02-01:
Hello, @CodeMonkeyX ,

We are sorry that we have still not implemented this Feature to PrusaSlicer. We start looking at that and we will try to implement chamber temps support in some of the next releases.  https://github.com/prusa3d/PrusaSlicer/issues/4356

CodeMonkeyX, 2024-02-02:
I will keep watching for it.

Thank you,

rongith, 2024-06-18:
I suppose the tip for inserting a global variable {chamber_temp} in filament "Start G-code" doesn't work for the same reason filament "Start G-code" is invoked after printer "Start G-code".
Then for now the only fallback solution is to use {filament_type}.
The following code inside filament "Start G-code" works but that doesn't help:
```
{global chamber_temp = 60;}
M141 S{chamber_temp}
```

rongith, 2024-09-13:
Chamber temperature is now available beginning with the 2.8 release.
It solves the issue for me but not for the OP who wants more custom variables like "Filament Color".

yonitjio, 2025-11-07:
@rongith I'm not the author of this issue.

closed 2026-09-29 by prusaslicer-bot[bot]