prusa3d/PrusaSlicer #10365: Windows 11 Crashes with bluescreen Cause igdkmdnd64.sys

blue0ceanby, 2023-04-15:
### Description of the bug

Windows 11 Crashes with bluescreen Cause igdkmdnd64.sys
When a model is loaded and ready to be sliced, after a few minutes

### Project file & How to reproduce

[20230415_215139.zip](https://github.com/prusa3d/PrusaSlicer/files/11240528/20230415_215139.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha6+win64

### Operating system

Windows 11 Prof

### Printer model

Geeetech A10

hakimio, 2023-04-16:
Seems to be Dell Inspiron issue: https://www.dell.com/community/Inspiron/Inspiron-7506-2-in-1-BSOD-igdkmdnd64-sys/m-p/8053012/highlight/true#M129407

blue0ceanby, 2023-04-16:
Thanks for the feedback will investigate. It is interesting that it only BSOD if a sliced model is loaded in the Prusa Slicer. I don't have any other BSOD with 3d programs like Freecad or Fusion.

kubispe1, 2023-04-18:
I discussed this issue with the developers. There could be several possible issues like mentioned driver or OS, but the application should not cause a BSOD error. I am closing this issue.
Thanks you both

closed 2023-04-18 by kubispe1