prusa3d/PrusaSlicer #10366: Slow Fast Travel when approaching supports or small objects

GadgetNutt, 2023-04-15:
**Is your feature request related to a problem? Please describe.**
I am printing at around 260mm/s and could print faster.  However, it keeps knocking down supports when approaching at that speed.

**Describe the solution you'd like**
I'd like the print head to slow down as it reaches and leaves supports and small objects to avoid damage and then continue at the defined higher travel speed.

**Describe how it would work**
I think I just did.

**Describe alternatives you've considered**
Not sure there are any?

GadgetNutt, 2023-04-19:
I tried z-hop.  It just made a horrible mess of my print..

kubispe1, 2023-05-25:
Could be similar issue... #10226

GadgetNutt, 2023-05-26:
I think you could be right.  It would make sense with how many failed prints I have had from it knocking the supports down.

closed 2026-09-29 by prusaslicer-bot[bot]