prusa3d/PrusaSlicer #10369: Slicing perimeter with travel speed

1Euro7Cent, 2023-04-16:
### Description of the bug

An inner perimeter is sliced with speeds of travel speed. Here is a video showing it:

https://user-images.githubusercontent.com/67194495/232299021-92df158d-bdea-4c0f-b78a-ecee0b77b7de.mp4

### Project file & How to reproduce

[bug report on slicing with wrong speed.zip](https://github.com/prusa3d/PrusaSlicer/files/11241945/bug.report.on.slicing.with.wrong.speed.zip)

see video above to see it

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha6+win64

### Operating system

Windows 10 64 bit

### Printer model

Sovol sv06 with klipper

kubispe1, 2023-04-19:
Thanks for the video and description. This issue is already fixed. Should be ok in the next release.
there it is: #10229

kubispe1, 2023-04-19:
Tested in PrusaSlicer-2.6.0-alpha6+91
[image]

closed 2023-04-19 by kubispe1