prusa3d/PrusaSlicer #10372: Organic support slicing causes "bad function call" error.

agrubb, 2023-04-16:
### Description of the bug

Trying to slice a relatively simple model with supports on two small areas.  Whether I use painted on supports or supports based on the overhang angle, the slicing fails when using the new organic supports.  The slicing operates fine for grid based supports or other methods.

For other models I have tried with the new organic supports, the slicing has worked fine.

When slicing, an error message is displayed only with:
```
PrusaSlicer has encountered an error

bad function call
```
[image]

### Project file & How to reproduce

[failed_organic_supports.zip](https://github.com/prusa3d/PrusaSlicer/files/11242570/failed_organic_supports.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha6+win64

### Operating system

Windows 10

### Printer model

Prusa Mini+

agrubb, 2023-04-16:
I have also tried repairing the STL and re-slicing, the same error is still shown.

kubispe1, 2023-04-17:
- I think it could be, duplicate issue of #9892
- should be fixed in new release by this commit: 23d74f453eafa90af65cd2bcf82360081fb6d2bd
- tested with PS 2.6.0 alpha6+83
there is sliced model:
[image]
- Closing the issue. In case of any problem let me know

closed 2023-04-17 by kubispe1