prusa3d/PrusaSlicer #10374: [improvement] Cut - dowel connectors placed flat by default for better endurance

KubaMiszcz, 2023-04-16:
**feature**
when I cut part and add connectors as dowels they are placed vertically with shear plane parallel to layer which is 
weakest plane

**Describe the solution you'd like**
place dowels rotate 90deg by Y (or X) so their shear plane is perpendicular to layer
it is even better when they are cut in half for better adhesion so for 3 dowels I have 6 half dowels
[image]

or dowels shape as square? hex?

foreachthing, 2023-04-16:
Which version of PS are you using?
In 2.6.0a6 I have several options for connectors:
[image]
Same for Plug and Dowel.

kubispe1, 2023-04-17:
@KubaMiszcz add original 3mf with issue. We can check it.

KubaMiszcz, 2023-04-20:
> Which version of PS are you using? In 2.6.0a6 I have several options for connectors: [image] Same for Plug and Dowel.

ah yes, I miss it, so when I choose square problem solved

closed 2026-09-29 by prusaslicer-bot[bot]