prusa3d/PrusaSlicer #10378: previous_extruder = -1 during first toolchange g code

bongo505, 2023-04-17:
### Description of the bug

I use custom G code for tool change on a dual head extruder. The slicer inserts the custom G code in the startup sequence, changing to T1, but the value of "previous_extruder" is -1 instead of 0
All subsequent calls to the toolchange code set previous_extruder to expected values.

ERROR:
[image]
OK:
[image]

Note that I had been fiddling around with ways to fix it, hence the conditional statement further down which is commented out. I also realise the initial M104 does not need T{previous_extruder}, and interestingly {temperature[previous_extruder]} actually returns the correct value here, and that I should have a M104 command for the new extruder, but I forgot that bit - it shouldn't affect the parsing of the expressions however.

Custom G Code:
G92 E0 ;Reset the extruder's origin/length
M104 **T{previous_extruder}** S{temperature[previous_extruder] - 50} ; reduce temp
G0 F5040 X{wipe_tower_x} Y{wipe_tower_y} ; force the thing to do a tool change over the centre of the wipe tower
T{next_extruder} ; change extruder
G92 E0 ;Reset the extruder's origin/length
M105 ; report temp
M109 S{temperature[next_extruder]} ;Wait for new extruder Hotend Temperature 
**NOTE NEXT LINE IS COMMENTED OUT**
;{if previous_extruder == -1}M104 T0 S{temperature[0] - 50}{else}M104 T{previous_extruder} S{temperature[previous_extruder] - 50}{endif} ; set the previous extruder to standby - the -50 is a wild stab here, some filaments may need different settings

### Project file & How to reproduce

[3D Print Parts.zip](https://github.com/prusa3d/PrusaSlicer/files/11248204/3D.Print.Parts.zip)

Don't need to do anything particular, just slice and export G code as usual.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.5.2+win64

### Operating system

Windows 10

### Printer model

Snapmaker A350 with Dual Extruder

Maycuz, 2023-05-19:
Same behaviour is seen on PrusaSlicer-2.6.0-beta2+win64

awabom, 2023-07-06:
Just a note on your formula for cooling down. Here's a formula which I think mimics the 'ooze prevention' feature of prusaslicer and checks filament "idle temperature" value and the print settings "temperature variation" value. Replace 'first_layer_temperature' with 'temperature' if needed :)

`M104 S{(is_nil(idle_temperature[0]) ? (first_layer_temperature[0] + standby_temperature_delta) : idle_temperature[0])} T0`

stonemandave, 2023-08-23:
Any progress on this I'm going through the same thing. -1 should return next extruder 1.

closed 2026-09-29 by prusaslicer-bot[bot]