prusa3d/PrusaSlicer #10380: glitch with fan speed display

Rattlhead, 2023-04-17:
### Description of the bug

In the settings and in the g code, 70% is registered, but for some reason 100% is in the display

### Project file & How to reproduce

[image]
[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 alpha 6 and 2.5.1

### Operating system

macos

### Printer model

ulti steel

kMik2002, 2023-04-18:
у вас указано минимальное время печати слоя 20 секунд и возможный максимум обдува 100%, соответственно, скорость вентилятора увеличивается до 100%

kubispe1, 2023-04-19:
It is strange and confusing. Please attach 3mf and generated G-Code. There could be a bug in the G-Code preview or there is some overwrite ??... Who knows only from the description.

Rattlhead, 2023-05-17:
[image]
in g code M106 S216.75, about 80% of the fan power is displayed, but in legend 100 is displayed%

Rattlhead, 2023-05-17:
[Архив.zip](https://github.com/prusa3d/PrusaSlicer/files/11495516/default.zip)
gcode and 3mf project

TravisWilder, 2023-05-30:
Still the same for me in B3

lukasmatena, 2023-06-13:
@Rattlhead Could you please set PrusaSlicer to English and check if the problem is still there? Thanks.

lukasmatena, 2023-06-13:
@Rattlhead Could you please also share the exact version of your OS?

bubnikv, 2023-06-13:
What's your operating system? MacOS Ventura?

lukasmatena, 2023-06-13:
@Rattlhead Thanks. Could you please give it a try in the following build? It is 2.6.0-rc1 plus [this commit](https://github.com/prusa3d/PrusaSlicer/commit/d492fc0efbff2758680f6ad310eb8dfd7c20c54c).

[PrusaSlicer-2.6.0-rc1+1-lm-fan-speed-bug-investigation-linux-x64-GTK3-gd492fc0-202306130717.AppImage](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.6.0-rc1+1-lm-fan-speed-bug-investigation-linux-x64-GTK3-gd492fc0-202306130717.AppImage)
[PrusaSlicer-2.6.0-rc1+1-lm-fan-speed-bug-investigation-MacOS-universal-gd492fc0ef-202306130729.dmg](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.6.0-rc1+1-lm-fan-speed-bug-investigation-MacOS-universal-gd492fc0ef-202306130729.dmg)
[PrusaSlicer-2.6.0-rc1+1-lm-fan-speed-bug-investigation-win64-gd492fc0ef-202306130718.zip](https://prusaslicer:slicer@slicerbuilds.prusa3d.com/files/lm/PrusaSlicer-2.6.0-rc1+1-lm-fan-speed-bug-investigation-win64-gd492fc0ef-202306130718.zip)

lukasmatena, 2023-06-14:
Should be fixed in [2.6.0-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-rc2). Closing.

closed 2023-06-14 by lukasmatena