prusa3d/PrusaSlicer #10381: Increase strength by offsetting layers

julie777, 2023-04-17:
**Is your feature request related to a problem? Please describe.**

3D printed parts are typically less strong than injection molded parts. This is inherent in the way Fused Deposition Manufacturing (FDM) models are printed, one layer at a time. The result is that for any two layers there is a weak point at the layer to layer bond and that weak spot always lines up all the way through the wall of the part being printed.

This is most obvious When creating parts with thin walls, for example a cup or a box. If there is an adhesion failure, it will be at the layer line. There are multiple techniques to improve layer adhesion (listed below) but they do not change the basic properties of FDM that cause the problem.

With machined parts, the weak point is a sharp corner where stress concentration is the highest. A chamfer or fillet is used to reduce the stress concentration at the corner and reduce the chance of failure.

With FDM, using a chamfer or fillet can make some parts stronger, but it does not address the problem of the layer joint going all the way through the part. In fact, adding a chamfer or fillet can increase the stress concentration at the layer where the fillet ends. In the image below, the fillet was used to make the wall to base connection stronger, but the failure point just moved up to the edge of the fillet. This is due to the layering of the model in FDM, which always causes a stepped transition from one layer to the next. With a machined part the fillet is a completely blended transition that completely removes the stress concentration.

Here is an example of separation at the edge of a fillet.
[image]

**Describe the solution you'd like**

Using a three line thick wall as an example, it should be possible to print the outer two lines at layer height n and the inner line at layer height n+.5. This prevents a single layer line that goes all the way through the wall creating a weak point. 

The concept is similar to a brick wall where the bricks are always offset horizontally to add strength by preventing a vertical "seam".

**Describe how it would work**

When the slicer detects that the model is transitioning to a thin wall, it can either over extrude or under extrude the center layer to start the offsetting process. From that point it is effectively changing the layer height to 1/2 the original value, but extruding the filament at full height. It alternates for each layer between the lines forming the walls.

Because the layer being printed is always higher than the previous layer there is no problem with the nozzle hitting the previous layer extrusion.

Here is a crude drawing of the current way a three perimeter wall gets printed on top the bottom layer.

[image]

You can see that the "squish" of the extrusion causes some overlap from layer to layer and between walls. You can also see how there can be internal gaps and there is always a horizontal layer line completely through the wall.

Here is a possible implementation of the shifted wall solution, using a half thick inner wall to get started.

[image]

If we ignore the bottom to wall interface, we can see how much stronger the wall will be because it no longer has any shear lines that go all the way through it. In addition, the squish during extrusion can completely eliminate voids without the risk of deforming the wall.

While I am suggesting this primarily as a way to strengthen thin walls, it could be use anywhere in a model to add strength.

**Describe alternatives you've considered**

- Adding more walls - uses more filament
- printing slower to allow more layer bonding
- higher filament temp and less cooling to melt the previous layer more for better bonding

The problem with all the existing solutions is that they do not get rid of the inherent problem caused by printing a layer at a time and having every layer line go all the way through the part.

**Additional context**
Add any other context or paint drawing about the feature request here.

neophyl, 2023-04-18:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/8621

vovodroid, 2023-04-19:
I added extrusion width variation on even layers in branch https://github.com/vovodroidprusa/PrusaSlicer/tree/variable-extr-width
So it doesn't shift layers, but shifts line borders, thus increasing strength.

NKT00, 2023-04-29:
You could even try having slightly higher and lower extrusion and/or fractionally shifted lines, intermittently on the inside walls, to give the next line of plastic better "keying" - much like a brick wall.

vovodroid, 2023-05-07:
> having slightly higher and lower extrusion and/or fractionally shifted lines, intermittently on the inside walls

It's too complicated for my knowledge of Prusa code ))).

twilkinson, 2024-02-29:
Has there been any progress on this requested feature?  I just saw this video (https://www.youtube.com/watch?v=5hGm6cubFVs) which gives a good description and references a feature in another slicer (Simplify3D) that facilitates it.

rob-miller, 2024-11-10:
Seems like a duplicate of #1823

closed 2026-09-29 by prusaslicer-bot[bot]