prusa3d/PrusaSlicer #10382: Slice/Wall merging issue

DVSVIDEO, 2023-04-17:
### Description of the bug

It slices incorrectly into nearby walls. It fuses, unites.
It does this even if you slice in classic mode and in Arahne mode.

### Project file & How to reproduce

Load the model
Slice it.
[Air_Spinner_2.zip](https://github.com/prusa3d/PrusaSlicer/files/11253974/Air_Spinner_2.zip)
[image]

This is the desired target (Orca Slicer):
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha6+win64-202303311519_signed

### Operating system

Win10

### Printer model

Micromake C1 modified core xz printer

neophyl, 2023-04-18:
I’m on mobile at the minute so can’t download and check your project but reduce the value in slice gap closing radius. This is the usual cause of this.

kubispe1, 2023-04-19:
I noticed also that XY compensation also influence this issue significantly. There is my test model. Should work for you.
[Air_Spinner_2_Edit.zip](https://github.com/prusa3d/PrusaSlicer/files/11272385/Air_Spinner_2_Edit.zip)