prusa3d/PrusaSlicer #10385: Divide by zero \ organic support

Rais93, 2023-04-18:
### Description of the bug

The app encounter an error "divide by zero" when slicing and asks me to restart the app. This only happens when i choose organic support, with or without painting. Seems unrelated to 
- two different sourced stl
- 02 or 0,3 layer height
- alpha 5 and alpha 6

### Project file & How to reproduce

[error_organic_files.zip](https://github.com/prusa3d/PrusaSlicer/files/11262811/error_organic_files.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 alpha6

### Operating system

win10

### Printer model

biqu hurakan with mainsail

kubispe1, 2023-04-19:
I am not able to reproduce this issue. Please can you attach .3mf? Could be valuable to test. We already fixed some of these issues. 
There could be also a duplicate: #9555

Rais93, 2023-04-19:
Yes it look like identical to #9555

Here the same STL converted into 3mf by fusion360
[Jethousing v1.zip](https://github.com/prusa3d/PrusaSlicer/files/11274021/Jethousing.v1.zip)

But i also tried to use a simple rhomboid with a steep angle and the problem persist if i choose Organic

kubispe1, 2023-04-20:
:D I expected 3mf from PS with settings that fail. Then I can inspect what kind of settings combination causes the error.

Rais93, 2023-04-20:
Got it.
[Jethousing-fromPS.zip](https://github.com/prusa3d/PrusaSlicer/files/11283672/Jethousing-fromPS.zip)

closed 2023-04-20 by kubispe1

Rais93, 2023-04-20:
what is SPE1451?

bubnikv, 2023-04-20:
It is our internal ticketing system.