prusa3d/PrusaSlicer #10389: Prusa Slicer 2.6.0-alpha6 crashes when opening a 3mf project that I saved with the same slicer.

juckendesAuge, 2023-04-18:
### Description of the bug

Hello,

I used Prusa Slicer "2.6.0-alpha6 (Appimage GTK3)" to enlarge a model to 300% and split the individual parts with the new slicing tool. I saved it yesterday as a 3mf project.
Today I can't open it anymore because the alpha version crashes.
However, you can open it with version 2.5.1 and 2.5.2 without crashing.

[repaired_alien-tail.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11265558/repaired_alien-tail.3mf.zip)

### Project file & How to reproduce

Open the 3mf project with "PrusaSlicer-2.6.0-alpha6+linux-x64-GTK3-202303311518.AppImage"

sha256sum:
c11830411e5426c4f2540d1a9ed2e9c8f2c87196f3cb6db04aaa288a9f9aa5e4  PrusaSlicer-2.6.0-alpha6+linux-x64-GTK3-202303311518.AppImage

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha6+linux-x64-GTK3-202303311518.AppImage

### Operating system

Fedora release 37 (Thirty Seven)

### Printer model

Prusa MK3S+ 0.6mm Nozzle

kubispe1, 2023-04-19:
Hi, thanks for the issue. 
- I can confirm that for Win it freezes (PrusaSlicer-2.6.0-alpha6+91) 
- For 2.5.2. it works OK
Could be a duplicate of : #10319

kubispe1, 2023-04-19:
We checked it and it is a duplicate of #10319
I am closing it. 
Thanks

closed 2023-04-19 by kubispe1