prusa3d/PrusaSlicer #10400: Version 2.6 Alpha 6, bridging angle seemingly has no effect

thwerks, 2023-04-20:
### Description of the bug

Setting "Bridging Angle" setting to any value has no effect. Bridge lines always oriented in same angle no matter the value

### Project file & How to reproduce

open any project with bridges, change Bridge Angle value.  Only solution appears to be to rotate the part on the the bed to different angle.

rename file to .3mf
[Body.txt](https://github.com/prusa3d/PrusaSlicer/files/11285692/Body.txt)

[image]

[image]

[image]

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0 alpha 6

### Operating system

Debian 11

### Printer model

Ender 3 Pro

scottmudge, 2023-04-20:
This was already reported multiple times. You have to set it in radians, not degrees. It will be fixed.

sarusani, 2023-04-20:
I'm pretty sure this happens because it's not bridging over "empty space" but bridging over infill. Don't know if it's intended behavior or not.
Bridging over "empty space" behaves correctly and takes the bridging angle into account.

thwerks, 2023-04-20:
> I'm pretty sure this happens because it's not bridging over "empty space" but bridging over infill. Don't know if it's intended behavior or not. Bridging over "empty space" behaves correctly and takes the bridging angle into account.

the first bridge does attach directly to the infill as expected and does a good job. However the second bridge layer directly attaches to the perimeter and barley touches infill at all, its bridging over thin air and the print fails repeatedly. However, by rotating the part 90 degrees on the bed, the first bridge layer stays the same, but the second or uppermost bridge layer rotates 90degress and chooses a much more prudent orientation for bridge lines and the print repeatedly succeeds. Intuitively it seems like the purpose of "Bridge Angle" is so that you dont need to rotate the part to correct bridging alignment problems.

kubispe1, 2023-04-21:
Hello! As you may have noticed, the 'Bridge angle' parameter doesn't control bridges over sparse infill. This parameter only affects the angle of standard bridges. The angle of bridges over sparse infill is determined automatically to follow the infill. We're currently fine-tuning the algorithm to anchor the infill and perimeter in the best way possible.

thwerks, 2023-04-21:
really appreciate the reply, hope this can make more sense in the near future. If you can see my explanation the bridging my case is rather ambiguous and inexplicable.  Thank you for your consideration.

b3n3d1k7, 2023-06-16:
> matically to follow the infill.

I'm trying to print a wide, but narrow part with 0 infill and the bridges to close the part use a 45° angle, which is clearly not the best option here ;-)
[image]

closed 2026-09-29 by prusaslicer-bot[bot]