prusa3d/PrusaSlicer #10404: Problems in opening multiple 3mf files

H-555666, 2023-04-20:
### Description of the bug

If you drag and drop 2 or more .3MF files (just model file, not project file) into prusaSlicer, it will just open one of them with no error messages.

### Project file & How to reproduce

1.find multiple .3MF files anywhere, make sure they are just 3Dmodel files.
2.drag and drop these files into your prusaSlicer.
3.you might find only one file has import in it.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6+win64

### Operating system

windows10

### Printer model

creality CR-5pro

sarusani, 2023-04-20:
Go to preferences and select "Show load project dialog".

H-555666, 2023-04-21:
> Go to preferences and select "Show load project dialog".

I've already selected it and when I drag and drop some .3MF files it will just show a dialog called loading and quickly disappeared with only one model loaded in it

closed 2026-09-29 by prusaslicer-bot[bot]