prusa3d/PrusaSlicer #10406: BUG thick bridge not disable

Victor-Ochoa, 2023-04-21:
### Description of the bug

Disabling the thick bridge has no effect on slicing in the alpha version, in the release version it works.

When I disable the 'thick bridge' checkbox in the release version it makes the 'bridge enfill' the same height. The bug occurs in the alpha version where slicing has the same behavior of increasing the height of the layer regardless of whether the checkbox is checked

prusa slicer 2.5.2  thick bridge disable
[image]

prusa slicer 2.6 alpha 6  thick bridge disable
[image]

### Project file & How to reproduce

1 - I'm use the calibration cube for test 
2 - disable thick bridge

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha 6

### Operating system

Windows 11 - 64

### Printer model

Ender 3 s1 Pro

kubispe1, 2023-04-21:
Hi, please check the Change log notes of  PrusaSlicer 2.6.0-alpha5, section **Extend sparse infill** here is a link : https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha5
> The bridge infill is now always using 'Thick bridges'.

closed 2023-04-21 by kubispe1

Victor-Ochoa, 2023-04-21:
@kubispe1 Could you tell me the reason for this decision? this greatly affects the quality of the parts because of the delay in stabilizing the flow causing external lines.