prusa3d/PrusaSlicer #10411: Dynamic Overhang Speed generates G1 F0

markusyeo, 2023-04-21:
### Description of the bug

[image]
When bridge overhang speed is set to 0%, it generates a gcode file with G1 F0, instead of using a default speed setting.

[image]

### Project file & How to reproduce

Used a file with bridges, it crashed in between the print due to G1 F0 which should not be allowed in the first place.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6

### Operating system

Windows 10

### Printer model

Sunlu T3

MandicReally, 2023-05-12:
Just commenting to say this happens for me as well.  

Expected result: Defaults to Bridging Speed set above Dynamic Overhang Speeds.  

Actaul Result: Sets speed to 0 and print fails.

Jan-Soustruznik, 2023-06-22:
Hello, @markusyeo , 

could you check with final release 2.6.0, https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0

I can not evoke reported problem or could you share a 3mf file wit this issue? 

@MandicReally  Could you share a 3mf file, please? I can not reproduce with final release 2.6.0

thank you, have a nice day

markusyeo, 2023-06-22:
Hello, it has been awhile since I've encountered this issue, but it was a result of no support as well as a model which had in bridge labelled parts. I'll check if this still occurs on the current stable release and update back

closed 2026-09-19 by markusyeo