prusa3d/PrusaSlicer #10416: Oversize model crashes app

Gaanon1, 2023-04-22:
### Description of the bug

PrusaSlicer 2.6 Alpha6, using a Mini+ build volume, when I placed a model that is too large to fit on the build plate and selected Auto Arrange with rotation enabled, PrusaSlicer stopped responding and the app closed.

### Project file & How to reproduce

[p3.zip](https://github.com/prusa3d/PrusaSlicer/files/11299938/p3.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.6.0 alpha 6 Win64

### Operating system

Windows 10

### Printer model

Prusa Mini+

kubispe1, 2023-04-24:
I can confirm a crash for 2.6.0 a6
Maybe it is the duplicate issue: #10278 
We are working on a fix. We can check tomorrow and let you know.
Thanks for the model.

Gaanon1, 2023-04-24:
I had the same crash on a different model that started oversized.  if I manually rotated the model it would fit (I'm printing it now).  However, auto arrange still crashes.
[p5.zip](https://github.com/prusa3d/PrusaSlicer/files/11315075/p5.zip)

kubispe1, 2023-05-17:
This crash should be fixed with PS 2.6.0-beta1. If any issues persist, please let us know. Thank you.

closed 2023-05-17 by kubispe1