prusa3d/PrusaSlicer #10417: PrusaSlicer 2.5 Causes Severe Layer Misalignment

sweetblackberry, 2023-04-22:
### Description of the bug

When printing a cylinder z wobble test model, its gcode file was sliced by PrusaSlicer 2.5 or 2.6, it has severe layer misalignment which is similar to the example pictures of Pressure Equalizer explanation in PrusaSlicer 2.5.0-alpha2 release.
However, both values of Pressure Equalizer are 0, I didn’t edit them at all.
When slicing the same cylinder model in Cura, the severe layer misalignment is completely gone.

I tested about 20 to 50 times and strongly feel it’s a bug of PrusaSlicer software. 

### Project file & How to reproduce

[image]
[image]
[image]
[image]
[image]
[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.2; 2.6.0-alpha4

### Operating system

Win10

### Printer model

Ender 3v2 belt Z

zswickliffe, 2023-04-22:
I believe this is the same issue as my submission. Can you confirm whether the slicer reverses directions where the layers are misaligned?

sweetblackberry, 2023-04-23:
> I believe this is the same issue as my submission. Can you confirm whether the slicer reverses directions where the layers are misaligned?
> 
> 
> 
> My issue here: https://github.com/prusa3d/PrusaSlicer/issues/10392

So far I don't know how to check this in PrusaSlicer, I mean layers seem to be vertically aligned but when printing it's misaligned.

Jan-Soustruznik, 2023-06-14:
Hello, @sweetblackberry , 

Could you share g-code and 3mf file, please?

closed 2026-09-29 by prusaslicer-bot[bot]