prusa3d/PrusaSlicer #10419: Layer change colour dialog opens off screen in 2.6

l0ne5tar, 2023-04-22:
### Description of the bug

When I try to right click edit the colour of a layer change in sliced/preview mode the modal colour selection dialog is opened off screen and to all intents and purposes the app has locked up. 

This does not occur with 2.5.

I suspect (but have not confirmed) that this is due to using this computer with high DPI scaling enabled (~250%, im using a TV :) )

here's the output from Nirsoft's WinLister:

with 2.5 the colour dialog opens at:

`Location: (1377, 603)	Size: (1452, 954)	`

and with 2.6 (alpha 6) it opens at:

`Location:  (5616, 2734)	Size: (1390, 1014)`

The screen is 3840 x 2160. 
(Incidentally, with DPI scaling at 250-300% the legend in the preview mode is really intrusive.)

sys info:
PrusaSlicer
Version:   2.6.0-alpha6+win64
Build:     PrusaSlicer-2.6.0-alpha6+win64-202303311519

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19044), 64-bit edition
Total RAM size [MB]: 68,641MB
OpenGL installation
GL version:   4.6.0
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 2080/PCIe/SSE2
GLSL version: 4.60.0

### Project file & How to reproduce

To repro:

- Set windows to dpi scaling of 300%
- Load prusa slicer 2.6a6
- load any model
- insert a colour change
- r-click a layer after the colour change to trigger the colour dialog
 - observe locked up application ( you can recover with alt f4 or external program (such as WinLister) to recenter the dialog)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha6+win64-202303311519

### Operating system

Windows 10 (build 19044), 64-bit edition

### Printer model

Mk3S

l0ne5tar, 2023-04-30:
I think this is the cause of 
https://github.com/prusa3d/PrusaSlicer/issues/10135

l0ne5tar, 2023-12-21:
Hi, this is still broken, how do we raise some attention?

YuSanka, 2023-12-21:
Thanks for your report. 
We can reproduce this issue.
SPE-2089

lukasmatena, 2024-02-09:
Fixed in [2.7.2-alpha2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.2-alpha2). Closing.

closed 2024-02-09 by lukasmatena