prusa3d/PrusaSlicer #10423: PrusaSlicer 2.5.2 problem with layer shift

Sihill96, 2023-04-22:
### Description of the bug

Layer shift occurred at random times. This did not occur in every printout. Changing the firmware to version 2.6 helped.

[image]
[image]
[image]

### Project file & How to reproduce

[E3DV6_OR_VOLC_40MM_CAP_V7.5.zip](https://github.com/prusa3d/PrusaSlicer/files/11302360/E3DV6_OR_VOLC_40MM_CAP_V7.5.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.2

### Operating system

Windows 10

### Printer model

Ender 3

tjhalva, 2023-04-23:
If you print the same gcode does the layer shift happen in the same location?

Here's some stuff I wrote about layer shifting:
https://github.com/prusa3d/PrusaSlicer/issues/8796

I never figured out specifically what my problem was, nor could I isolate it to something trivially detectable. It seems to occur when there is a sudden change in the direction of the Y axis (the table moves quickly). On my printer there is an audible "thump" when a layer shift occurs.

DaGwhite, 2024-10-28:
I've run into a very similar issue.  It's a varying Y-axis shift that depends on the Y position of the model.  The shift varies in size and direction.  It appears to be triggered by some embedded text in the wall of the model, and the shift occurs at the bottom of the text.  I did all sorts of tests, plotted the shift vs Y position, and took some photos.  Rather than try to put it all in this post, I've attached a PDF with all the details.  There is also a Zip file with the PrusaSlicer file, the .STL file, and a copy of the PDF.  I think this may also be related to the “buldge/bulge” issue that has over 20 pages of posts on the PrusaSlicer user forum:  https://forum.prusa3d.com/?wpfs=buldge.

[Prusa Slicer Y Shift Bug.zip](https://github.com/user-attachments/files/17547356/Prusa.Slicer.Y.Shift.Bug.zip)
[Y Shift Bug.pdf](https://github.com/user-attachments/files/17547357/Y.Shift.Bug.pdf)

closed 2026-09-29 by prusaslicer-bot[bot]