prusa3d/PrusaSlicer #10424: Non-Sequential Z-Height Printing

tjhalva, 2023-04-23:
I don't really know what to name this, but the premise is if you have an object, like an "H", instead of constantly move from one vertical side of the "H" to the other and potentially creating stringing and doing excess travel, to define some acceptable "z" height that would be printed on "one side" before traveling to the other.

**Is your feature request related to a problem? Please describe.**
Reducing travel and stringing when building tower like elements that are generally far apart.

**Describe the solution you'd like**
Build upward in the z-direction to some defined threshold, and then move in the x-y plane and return to a lower z-height.

**Describe how it would work**
* Define some z-height distance as the threshold for growing upward on a single feature.
* Detect when the printer head is smaller than the distance between the towers.

**Describe alternatives you've considered**
None

**Additional context**

This is how an "H" is printed today during monotonic z-height increases:

```
5----->6
4<-----3
56789012
4<-----3
1----->2
```

Instead, the left and right sides could be printed multiple z-layers at to minimize travel:

```
4--    6
3  \-->5
21098765
2--    4
1  \-->3
```

Tupson444, 2023-04-23:
This has been suggested before (don't remember which issue), but we definitely need this. At least to be able to print a couple of mm before moving to the next "island".

neophyl, 2023-04-23:
Was this one I think https://github.com/prusa3d/PrusaSlicer/issues/7202

closed 2026-09-29 by prusaslicer-bot[bot]