prusa3d/PrusaSlicer #10425: Interwoven Perimeters

tjhalva, 2023-04-23:
Suppose there are 5 vertical perimeters and 5 top perimeters, a simple illustration of the current behavior is below:

```
VVVVVTTTTTT...
VVVVVTTTTTT...
VVVVVTTTTTT...
VVVVVTTTTTT...
VVVVVTTTTTT...
VVVVVIIIIII...
VVVVVIIIIII...
`````

* V = Vertical Perimeter Row
* T = Top Permimeter Row
* I = Infill

Interweaving the Top and Vertical perimeters in some fashion such as below may have some benefit to strength:

```
VVVVVTTTTTT...
VVVTTTTTTTT... <---
VVVVVTTTTTT...
VVVTTTTTTTT... <---
VVVVVTTTTTT...
VVVVVIIIIII...
VVVVVIIIIII...
`````
The layers with the `<---` have the Vertical Perimeters only 3 wide, while the Top Perimeter extends.

**Is your feature request related to a problem? Please describe.**
Eliminating the edge between the Vertical Perimeter and the Top Perimeter can create a stronger join.

**Describe the solution you'd like**
I'm not sure what patterns would be the most desirable, but the alternating concept illustrated above could simply have a checkbox for:
* Overlap Top Perimeter and Vertical Perimeter: True/False
* Overlap Width: Probably defined in mm, but there is some nuance with the thickness when using Arachne.

**Describe how it would work**
This is an optional feature that could be enabled when the Top and Vertical Perimeter settings are above some defined threshold. This doesn't make sense to do if they are 2 units, but 5 in the above example allowed for the box tooth pattern.

**Describe alternatives you've considered**
None.

**Additional context**
This was sort of inspired by the more complicated concept of staggering layers like a brick pattern to improve strength. That concept is much more complicated.

neophyl, 2023-04-23:
I might be missing something but you haven't defined what the difference is between a vertical perimeter or a top perimeter ? At the moment we have external perimeters and perimeters.  We don't have top or vertical, at least not in any definition I have seen.  So what exactly are you defining or referring to ?

tjhalva, 2023-04-24:
My terminology is all wrong. :(

"Top Perimeter" would be some combination of:
* Bridge infill
* Solid infill
* Top solid infill

"Vertical Perimeter" would be:
* External Perimeter
* Perimeter

"Infill" was intended to be "Internal Infill".

Here's the sequence from lower layer to top layer:

[image]
[image]
[image]
[image]

In the blue hand drawn oval, instead of always having the top meet the side, instead have the side extend a few layers deep.

tjhalva, 2023-04-24:
This probably makes the most sense to do on the "solid infill" layers.

```
OYYRRRRR...
OYYPPPPP...
OYYPPPPP...
OYYPPPPP...
OYYBBBBB...
```

Could have the purple cover like two rows less:

```
OYYRRRRR...
OYYPPPPP...
OYYYYPPP... <----
OYYPPPPP...
OYYBBBBB...
```

Each character is just the color of the layer in the above screens.

closed 2026-09-29 by prusaslicer-bot[bot]