prusa3d/PrusaSlicer #10426: missing layers in vase mode

jjz015, 2023-04-23:
### Description of the bug

When trying to slice airfoils in vase mode, there are periodic layer gaps throughout the slicing preview. 
[image]

This issue only appears once fully sliced, as the preview looks ok when at 90% slicing
[image]

### Project file & How to reproduce

[E423 v8 projectfile.zip](https://github.com/prusa3d/PrusaSlicer/files/11302880/E423.v8.projectfile.zip)
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha5

### Operating system

Windows 10

### Printer model

Ender 3

TTN-, 2024-01-13:
Having the exact issue here. Zooming in on a layer where there is a missing layer we see the last move is this:

[image]

Then ok back pedalling a little, see the leadup to that is this: 

[image]

And then it appears to trap itself. Like it's a path planning bug, or perhaps it's to do with the STL resolution. It's a 12MB STEP file that's just imported.

Here we can see there are 9 occurrences of the issue:

[image]

I did note someone changing the Slice gap closing radius in a video: 

[image]

From that default to, 0.012mm. Which reduceds the amount of missing layers to just 1. I am using a 0.6mm nozzle though, so ok lets go smaller: 0.005mm and the issue is entirely resolved, all layers are present

Using Prusaslicer-2.7.1

Glaedr304, 2025-04-26:
Adjusting the slice gap closing radius to 0.005 also fixed this problem for me as well

closed 2026-09-29 by prusaslicer-bot[bot]