prusa3d/PrusaSlicer #10427: crash not detected or stopped in sequential print

TeslaCT, 2023-04-23:
### Description of the bug

PrusaSlicer 2.6.0 alpha 6 did not alert to a crash for a sequential print. I sliced the model and it was printing. only last minute did I notice that the second part was below the first part and would require the gantry to pass through the first part. 

### Project file & How to reproduce

[FAB365_Starship_report_crash.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11302933/FAB365_Starship_report_crash.3mf.zip)
[FAB365_Starship_report_crash_0.6n_0.3mm_PLA_MINI_1h9m.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/11302940/FAB365_Starship_report_crash_0.6n_0.3mm_PLA_MINI_1h9m.gcode.zip)
[image]

sliced in prusaslicer 2.6.0 alpha 6 on MacOS arm64. sliced models Starship_cable and starship_booster. auto arranged on build plate. already had sequential printing turned on. did not show outline of needed keep out area or say that something was too tall. 
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha6+MacOS-arm64

### Operating system

  System Version:	macOS 13.3.1 (22E261)   Kernel Version:	Darwin 22.4.0

### Printer model

Prusa mini

sarusani, 2023-04-23:
PrusaSlicer will not detect all possible crashes. Please check out:
https://help.prusa3d.com/article/sequential-printing_124589

TeslaCT, 2023-04-23:
> PrusaSlicer will not detect all possible crashes. Please check out:
> https://help.prusa3d.com/article/sequential-printing_124589

agreed. but between 2.5 and 2.6 the behaviors are very different. I have the same(default) settings for extruder clearance. 2.5 won't let me slice the models in the same position but 2.6 will. 2.6 will also let me do sequential vase modes prints with color swaps. I haven't printed these but it does generate the gcode. which is not a behavior that worked before in 2.5.2
[Complete Vase Rose Remix_sequential_0.6n_0.4mm_PLA_MINI_2h19m.gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/11304565/Complete.Vase.Rose.Remix_sequential_0.6n_0.4mm_PLA_MINI_2h19m.gcode.zip)

rtyr, 2023-04-24:
Similar to https://github.com/prusa3d/PrusaSlicer/issues/10312. It was already fixed for next release.

TeslaCT, 2023-04-24:
understood. I forgot to search for other issues prior to posting. those other issues look like they will cover the changes.

lukasmatena, 2025-02-18:
Arrange and crash detection for sequential printing was improved in [PrusaSlicer 2.9.1-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.1-alpha1). I believe it solves this issue. Closing.