prusa3d/PrusaSlicer #10428: Perspective and Ortographic Option under "View" menu

sarvenn, 2023-04-23:
Sometime I need to switch between perspective and orto view modes. Currently, I need to go to click on configurations manu, select Preferences, go to Camera tab and tick/un-tick use perspective camera. This has a lot of steps.
So I suggest to add this under View menu, therefore it can be switched quickly.
[image]

foreachthing, 2023-04-23:
Also a hotkey would be nice.

Tupson444, 2023-04-23:
There already is a shortcut: K (at least on 2.5.0)

sarvenn, 2023-04-23:
> There already is a shortcut: K (at least on 2.5.0)

Yes there is but it could be also nice to have it under view menu. I think this would not be a big deal to accomplish.

closed 2026-09-29 by prusaslicer-bot[bot]