prusa3d/PrusaSlicer #10432: Organic Support Bug Prusaslicer 2.6 - alpha 6

juckendesAuge, 2023-04-23:
### Description of the bug

Organic supports are printed in the air, within a model.
The Left Organic Support is useless, it supports nothing, but is generated and printed.
Top Interface Layers are printed at a few organic tails diagonally. Wouldn't it be better to print a bridge first and have the organic supports automatically generated so that the corners of the bridge are supported?
The "Top Interface Layers" should then be printed on top of this.

Translated with www.DeepL.com/Translator (free version)
[image]
[image]
[image]
[image]
[22mm Klemme.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11304288/22mm.Klemme.3mf.zip)

### Project file & How to reproduce

Open the 3mf Projectfile in the zip.
Slice it.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha6+linux-x64-GTK3-202303311518.AppImage

### Operating system

Fedora release 37 (Thirty Seven)

### Printer model

Prusa MK3S+ 0.6mm Nozzle

neophyl, 2023-04-23:
Looks like a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/9601

Jan-Soustruznik, 2023-06-22:
Hello, @juckendesAuge, 

After modification of the Organic supports in b1/b2 most of reported issues above should be fixed or limited. But the issues with organic supports still remains in some cases. I will close this issue as duplicate and we will continue in the main topic https://github.com/prusa3d/PrusaSlicer/issues/9601. Thank you for your report.

Have a nice day.

closed 2023-06-22 by Jan-Soustruznik