prusa3d/PrusaSlicer #10433: Gap in substrate when thin writing is over - 2.6-alpha6

iuliux, 2023-04-23:
### Description of the bug

Incomplete layer that causes printing in mid-air on this particular case of very thin lettering.

[image]

[image]

### Project file & How to reproduce

Attaching the 3MF file to reproduce.

[measure_Back.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/11304332/measure_Back.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.6.0-alpha6+MacOS-universal

### Operating system

macOS 13.0

### Printer model

Mk3s+

iuliux, 2023-04-23:
Note: not reproducing on 2.5.2

kubispe1, 2023-04-24:
Hi, thanks for the issue. It is a similar issue to [#10371](https://github.com/prusa3d/PrusaSlicer/issues/10371)
This issue looks already fixed. I tested with the last master: PrusaSlicer-2.6.0-alpha6+125
The fix will be available in the next release. If you find some cases with problems let us know. Thanks
[image]