prusa3d/PrusaSlicer #10435: Add support base layer parameter

Havoc340, 2023-04-23:
Hello! 

I'm a long time s3d user trying to branch out, and there is one thing I miss a lot when it comes to quality of life. Adjustable support layers from the buildplate. 

I would like to see an option to change the number of layers under supports from the buildplate. 

I just had another support base layer rammed under my fingernail as I was trying to pluck it off without damaging my PEI bed. I figured it was time to do something about it. So I am making a request haha. 

Single layer support bases from the bed work well for adheasion and for the fuctionality they work great as they are, until it's time to remove them from the bed. They are so thin they just flex with the bed and stay firmly attached. They are also not rigid enough to pop themselves off the buildplate when cooled. 

Having the option to change the support base layer thickness from the buildplate would make the leftover support islands thicker and a little more likely to pop off the build plate when flexed or cooled. Or at the very least, make them thick enough to be knocked off the buildplate without needing to use a sharp metal tool like a thin scraper that could damage my buildplate.

neophyl, 2023-04-23:
Duplicate of many existing issues , for example https://github.com/prusa3d/PrusaSlicer/issues/9448

Havoc340, 2023-04-23:
> Duplicate of many existing issues , for example #9448

My bad. Well I'm glad to see I'm not alone haha. 

New to this github stuff.

closed 2023-04-23 by Havoc340