prusa3d/PrusaSlicer #10437: Custom Settings and Presets for different Nozzles and or Nozzle Materials

jDavidnet, 2023-04-24:
**Is your feature request related to a problem? Please describe.**
One part of swapping Nozzles is getting the printer and gcode files tuned for each Nozzle's thermal characteristics.  

As 3d printers get more advanced and nozzle swapping is getting easier physically, it would be nice to make swapping nozzles and dialing-in settings easier.  We have default settings for prints, filaments, and printers, but nozzles are more than just a diameter.  Nozzles have different thermal properties, wear, hardness, and slickness factors that you might want warnings or adjustments to your print with.

For instance, recently, I've been trying to get Prusament Galaxy Purple to print reliably on an MK3S+ with a modified V6 hotend with a copper block, and a Nozzle X .4mm Nozzle, but I kept getting clogged.  I also noticed that once I realized it was actually a hardened steel nozzle so, I increased the temperature to 230 C, and bingo, mostly flawless prints at .2mm & .3mm layer heights.  However, when I tried to print an older adaptive layer height gcode file, that's when it started clogging again. 

**Describe the solution you'd like**
I think it would be great to have a few more settings than just Nozzle diameter, maybe diameter, material, nozzle temperature delta, and hardness rating at a minimum.

Eventually, in the slicer, in addition, to drop downs for print, printer, and filament, you would have a nozzle dropdown, this would separate the speed from the nozzle selection and also allow mostly the same speed settings for a .4mm print, but would allow you to quickly select between a .4mm brass nozzle, .4mm TC, .4mm ObXidian X, .4mm Steel, ruby, diamond, etc....  Since most .4mm prints should be similar with only a few nozzle differences between them.

As a version zero of the feature, maybe add a temperature offset near the nozzle diameter under 
Printer Settings->Extruder [x]->

Then change the header "Size" to "Nozzle."
and then under the header, you'd have a few settings
Nozzle Size        [  ]mm
Nozzle Material  [  ] <- dropdown
Nozzle Temp offset +[  ]

Since printers are getting more flexible at the print head, more flexibility for the nozzle, extruder, and hot end are required.
Possible scenarios to consider in the long term are:
* Prusa Nextruder XL / MK4 Quick Swap Nozzle/ hot end
* Revo 6 system
* Wham Bam - Mutant 2 - Universal Tool System

And these material types for nozzles:
* Brass
* Steel
* Tungsten
* Tungsten Carbide
* Diamond Tipped - Brass
* Ruby Tipped - Brass
* ObXsidian - Coated Steel Nozzle with Copper Carrier
* Nozzle X - Coated Hardened Steel
* Copper
* etc...

Mr3Deee, 2025-05-03:
Yes, I would really like that too.

I would be even happy if I could change in the printer a per head temperature offset (ObXidian / Brass / Diamond) which would automatically be applied to a stored .3mf used in the slicer. 

Having multiple print heads, filament materials, nozzles (sizes, temps), etc. makes the re-print already an interesting exercise.

Adding a general +15 for ObXidian would be SUPER AWESOME and sounds like a small thing to do? Yes please?

Mr3Deee, 2025-05-11:
Hmm. Ok - the XL has a setting for hardened Nozzle (must have been added a while ago - mea culpa) however - I was not able to see how that affects the temperature print? According to the printer the temperature seems to be the same as before and I was not able to find a description on what it actually does.

Furthermore - using also Diamondback Nozzles - they are also hardened, but they do not need the temperature adjustments. 

So - what exactly does the setting do, and for what kind of nozzle is that meant? Any help would be greatly appreciated!

Mr3Deee, 2026-09-19:
I was speaking with more folks and they do not know what the setting "Hardened Nozzle" (in the printer) actually does.

Inside the slicer there are the "presets" and they explode. All kinds of nozzles, materials, and so on gives a great combination of profiles. Even worse - you load an external 3mf model and boom you have more settings.

It would be much better to have a setting for e.g. Prusa PLA and inside the setting you can manage different nozzle types (hardened, Diamond, standard..) which would do their temperature adjustments, nozzle size - modifying flow size, etc. Without this it simply becomes unmanagable.

Thanks for listening.