prusa3d/PrusaSlicer #10438: Random crash (with oversized model)

the-ress, 2023-04-24:
### Description of the bug

The application crashed unexpectedly with this crash dump:
[prusa-slicer.exe.19304.zip](https://github.com/prusa3d/PrusaSlicer/files/11312182/prusa-slicer.exe.19304.zip)

### Project file & How to reproduce

I added the following `Base/Frame - 5x6 - ZackFreedman.stl` from https://www.printables.com/model/242711-gridfinity-master-collection/files (by clicking the `prusaslicer://` link)
Then I noticed it was too big for my printer and mindlessly pressed Ctrl+R or something. Then PrusaSlicer crashed.

I was using default Prusament PETG @ 0.2mm on MK3S without any custom settings.
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.6.0-alpha6+win64

### Operating system

Windows 11 22H2 (build 22621.1631)

### Printer model

Prusa MK3S+

the-ress, 2023-04-24:
[prusa-slicer.exe.19296.zip](https://github.com/prusa3d/PrusaSlicer/files/11312719/prusa-slicer.exe.19296.zip)

Another crash with the same stacktrace - this time I pressed "A". (I think I pressed "A" instead of Ctrl+R before too.)

the-ress, 2023-04-24:
I found consistent repro steps:
1. Open a new instance
2. Drop this STL: [frame-5x6-zackfreedman.zip](https://github.com/prusa3d/PrusaSlicer/files/11312789/frame-5x6-zackfreedman.zip)
3. Press A
4. PrusaSlicer crashes

effndc, 2023-04-25:
I came to report the same, I am on Apple Silicon Mac.  Open [this model](https://www.printables.com/model/100454-watering-can/) and attempt to use Arrange function.

```
PrusaSlicer
Version:   2.6.0-alpha6+MacOS-arm64
Build:     PrusaSlicer-2.6.0-alpha6+MacOS-arm64-202303311518

Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Ventura Version 13.3.1 (Build 22E261)
Total RAM size [MB]: 34,360MB
OpenGL installation
GL version:   4.1.0
Profile:      Core
Vendor:       Apple
Renderer:     Apple M1 Pro
GLSL version: 4.10.0
Textures compression:       Enabled
<details>
<summary>Installed extensions:</summary>
GL_APPLE_client_storage
GL_APPLE_container_object_shareable
GL_APPLE_flush_render
GL_APPLE_rgb_422
GL_APPLE_row_bytes
GL_APPLE_texture_range
GL_ARB_ES2_compatibility
GL_ARB_blend_func_extended
GL_ARB_draw_buffers_blend
GL_ARB_draw_indirect
GL_ARB_explicit_attrib_location
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_occlusion_query2
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_separate_shader_objects
GL_ARB_shader_bit_encoding
GL_ARB_shader_subroutine
GL_ARB_shading_language_include
GL_ARB_tessellation_shader
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_cube_map_array
GL_ARB_texture_gather
GL_ARB_texture_query_lod
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_storage
GL_ARB_texture_swizzle
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_EXT_debug_label
GL_EXT_debug_marker
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_texture_compression_s3tc
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_sRGB_decode
GL_NV_texture_barrier
</details>
```

Jan-Soustruznik, 2023-04-25:
It seems that the problem is similar with https://github.com/prusa3d/PrusaSlicer/issues/10278

Closed issue as duplicate

closed 2023-04-25 by Jan-Soustruznik