prusa3d/PrusaSlicer #10439: overlapping support

RonaldVEss, 2023-04-24:
### Description of the bug

In making multiples of the same file i had a problem with the new organic supports. If the 2 models are to close the support of the other model will go into the model next to it as if it isn't there
[image]
[image]
[image]

### Project file & How to reproduce

[overlapping support.zip](https://github.com/prusa3d/PrusaSlicer/files/11313491/overlapping.support.zip)

slice the file as thay stand and check the lowest layers. 
this placement is the automatic placement from the slicer

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.6.0-alpha6+win64

### Operating system

Windows 10

### Printer model

Prusa mini

foreachthing, 2023-04-24:
Set as separate objects first, then merge 'em again.
[image]

dup: https://github.com/prusa3d/PrusaSlicer/issues/316 and many more...

kubispe1, 2023-04-24:
@foreachthing thanks for the advice. 
Instances have the same problems as part of this case. https://github.com/prusa3d/PrusaSlicer/issues/10376#issuecomment-1512108759
Closing this issue.

closed 2023-04-24 by kubispe1